autowarefoundation/autoware-projects #11: Metric calculation node development

mitsudome-r, 2022-02-23:
## Description
(Copied issue from https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/1446)
To implement and upload to AutowreAuto repository sample ROS2 node for metric calculation. As a sample, the calculation of deviation from planned path should be calculated.

### Purpose
To validate specific functionalities of AutowareAuto

### Desired Behavior
Metric calculation node subscribes to ROS2  topics published by Autoware or Simulation pipeline and publishes value of calculated metric

### Definition of Done  
- [ ] Sample metric calculation node developed and uploaded to the AutowareAuto repository
- [ ] Instruction how to run, and how to develop other nodes for calculation of different metrics

WJaworskiRobotec, 2022-04-20:
Similar feature related to trajectory evaluation : https://github.com/autowarefoundation/autoware.universe/pull/288

djargot, 2022-04-27:
Currently working on this issue. 
I am reusing parts of the planning_evaluator and motion_evaluator and expanding it to calculate various metrics, like:
* average velocity
* calculation of localization errors: lateral, absolute deviations from the trajectory
* heading angle error
* distance to closest obstacle

The final goals are:
* to have a generic metric calculator that more specific metric calculators (planning, localization, object detection, possibly more?) could inherit from.
* to make it easy to add new metrics under different categories (planning, localization, object detection, etc.)

djargot, 2022-05-24:
Recently added 2 separate evaluators (localization and kinematic), cleaned the code, and added basic unit tests.

I am currently trying to find the best way to find proper beginning and ending times between which metrics should be collected as it should happen only when a vehicle is driving. 
Additionally, working on segmentation evaluator and extracting common parts of all evaluators to parent evaluator.

WJaworskiRobotec, 2022-07-19:
https://user-images.githubusercontent.com/85557834/179841523-d43e5197-1bf0-40e1-85b8-cd0e51bf5d90.mp4

djargot, 2022-08-17:
I created initial version of segmentation metrics (https://github.com/autowarefoundation/autoware.universe/pull/1600) to calculate precision, recall, accuracy, and confusion matrix. Values are calculated both for a single point cloud as well as across whole scenario/ROS bag. 

I tested metrics using labeled point cloud provided by MORAI (related issue: https://github.com/autowarefoundation/autoware.universe/issues/669).

Example results for ground segmentation metrics for a single point cloud:
[image]

mitsudome-r, 2024-04-22:
Close due to stale issue

closed 2024-04-22 by mitsudome-r