autowarefoundation/autoware-projects #8: Open AD Kit CI/CD pipeline - Github

mitsudome-r, 2022-02-23:
## Description
Github based CI/CD that is hosted on AWS. This is a followup of: https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/1353 proposed for after the initial 1.0.0 release.

### Purpose

Cloud based continuous integration and deployment (see the discussion topics proposed by Kasper and Tanzawa from ASWG Meeting https://gitlab.com/autowarefoundation/autoware-foundation/-/issues/166)
- step 1: single container image (completed in #1353)
- step 2: multiple container images
- step 3: multiple container images + k3s

Step 2 and step 3 require configuration changes in the CI/CD which must be designed and implemented in Github.

### Desired Behavior

CI/CD pipeline for Autoware.Auto running in AWS

### Definition of Done  

1. [ ] Presentation at Open AD Kit WG outlining Github CI/CD plans
2. [ ] Autoware CI/CD Pipeline on multiple container images (step 2)
3. [ ] Autoware CI/CD Pipeline on multiple container images on Edge Workload + k3s (step 3)

LalithVipulananthan, 2022-09-13:
@kasperornmeck My understanding is that Ito-san was able to confirm that Autoware Universe can be run in multiple (monolithic) containers orchestrated by K3s, so for CI/CD what's the next step?

kasperornmeck, 2022-09-13:
@LalithVipulananthan 
When this issue was created, the idea wasn't to run a single container multiple times. We now know that this is a more difficult task than expected. I don't know if the TSC has made a decision how to proceed, i.e., if we wish to use Bloom to release ROS packages, see https://docs.google.com/presentation/d/1ICa6bl1q38jR9peueNOL0ycyT16U9M4RKD-pM-pLbGU/edit?usp=sharing.

LalithVipulananthan, 2022-09-15:
Setting to "Blocked" until there's a decision from the TSC regarding the Autoware release proposal mentioned in Kasper's comment above.

kasperornmeck, 2023-01-20:
@BonoloAWF - I suggest we close this issue as it won't go anywhere and the issues this issue tries to solve is addressed in the DevOps Dojos.

closed 2023-01-22 by BonoloAWF