autowarefoundation/autoware-projects #13: Simulator selection - define benchmarking setup

WJaworskiRobotec, 2022-03-29:


sglee-morai, 2022-04-06:
Here are two candidates for the path. If you find the first path is rather difficult to drive manually (since we don't need to drive automatically for the benchmark), the second one would be an easier choice. 

## The 1st Path
See the red line
[image]
Starting Point 
[image]
End Point 
[image]

##The 2nd Path 
See the red line
[image]
Starting Point 
[image]
End Point 
[image]

WJaworskiRobotec, 2022-04-07:
Thank you very much @sglee-morai . Let's test Path 1 it's not really difficult for manual driving