autowarefoundation/autoware-projects #21: Define Test Procedures to check Simulator's Compatibility with Autoware

sglee-morai, 2022-04-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

To define test procedures (e.g. test codes) to check whether a certain simulator is compatible with Autoware

Needs to define
1) what feature of Autoware should be able to integrate with a certain simulator
2) requirement for the simulator (e.g. a simulator should receive autoware_auto_msgs/VehicleControlCommand and the vehicle model within the simulator should respond accordingly.)
3) test cases for the requirements
4) test codes

### Purpose

To ensure that any simulator is compatible with the latest version of Autoware
Ultimately, streamlining the development process of Autoware-compatible simulators

### Possible approaches

A very simple method to compatibility with Autoware is included in this documentation: [https://morai.atlassian.net/wiki/external/1107034224/NmI1NGUxNmQzNmYxNDExZTk4NTYxYmQ0MzExOTI5MTA?atlOrigin=eyJpIjoiNWQyZTYwYTFiNGJkNGFjMGE5ZWMwZTdhNmYxZTczNmYiLCJwIjoiYyJ9](url)

Discussion needed: how to automate the tests or at least do in a systematic way.

### Definition of done

(Details Needs to be discussed)

sglee-morai, 2022-05-10:
### Proceed in two steps:
1) creating (mostly) manual procedures
2) automate the procedures

### Definition of Compatibility
-

sglee-morai, 2022-05-10:
## Test Procedures to test interfaces between a simulator and Autoware

### Requirement overview
link for the interface requirement: [link](https://github.com/orgs/autowarefoundation/discussions/234#discussioncomment-2638205)

sglee-morai, 2022-05-25:
# Update 22-05-25
- The initial version of the test codes for No. 1, 2, 3 is written

Discussion needed
- Create PR for what's written for now, and then work on the others?

sglee-morai, 2022-06-08:
#  Update 22-06-08
Test code for No.1 to 6 is written (Already some automation features are already implemented)

PR has been delayed since internally we have found things to refactor first, which Soohyeok (MORAI) is doing some refactoring.

WJaworskiRobotec, 2022-07-05:
PR with the code created here : [link](https://github.com/autowarefoundation/autoware.universe/pull/1107)