autowarefoundation/autoware-projects #25: Data recording minimum requirements

LalithVipulananthan, 2022-05-02:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

List the minimum set of ROS topics that should be recorded in order to enable reproduction of errors that occur during field testing

### Purpose

In order to fix Autoware errors or failures that occur during field testing, a minimum data set must be defined. Doing so will enable developers to:
- Calculate how much data is required per minute of field testing 
- Determine the minimum required storage space for their requirement (eg: 250MB of data is recorded per second, so a 1TB drive is necessary to record an hour's worth of data -> (250MB* 60 * 60) / 1000 = 900GB)

A minimum data set will also provide a foundation for developing functionality to automatically record data when field testing on time or maximum file size requirement (eg: record for one hour or until 1TB of storage is used up, then overwrite the previous ROSBAG file).

### Possible approaches

Although in theory, it should be possible to reproduce an error by recording all raw sensor and vehicle data, ROS 2 is not deterministic and so this is not guaranteed. One possible approach is to only replay processed input data into the specific part of the pipeline where an error has occured.

eg: to try and reproduce a planning error, only replay the relevant input data into the Planning module

### Definition of done

- TODO

### Blocked by

- Issue is blocked by the ongoing interface discussion that is to be finalised in the Autoware Architecture WG (TODO: add any specific issues relating to that discussion)
- The major interface is expected to be defined by May, so should be able to provide a list by early June

LalithVipulananthan, 2022-05-31:
@mitsudome-r Would it be possible to provide any links related to the interface discussion so that we can track progress of that to better understand when work on this issue can continue?

mitsudome-r, 2022-08-03:
@LalithVipulananthan 
Here are the minutes from the Architecture WG discussions: https://github.com/autowarefoundation/autoware/discussions?discussions_q=label%3Aarchitecture_wg

> Although in theory, it should be possible to reproduce an error by recording all raw sensor and vehicle data, ROS 2 is not deterministic and so this is not guaranteed. One possible approach is to only replay processed input data into the specific part of the pipeline where an error has occured.

At least we can start with minimal raw sensor data, and other primary data. This includes the following topics:
| Source  | topic name | message type | 
| ---          |  ---              |   ---                  | 
| Lidar     |  /sensing/lidar/{lidar_name}/pointcloud_raw | sensor_msgs/PointCloud2 |
| Camera | /sensing/camera/{camera_name}/image_raw | sensor_msgs/Image |
| Camera | /sensing/camera/{camera_name}/camera_info | sensor_msgs/CameraInfo |
| GNSS    | /sensing/gnss/{gnss_name}/nav_sat_fix | sensor_msgs/NavSatFix | 
| GNSS    | /sensing/gnss/{gnss_name}/orientation | [autoware_sensing_msgs/GnssInsOrientationStamped.msg](https://github.com/autowarefoundation/autoware_msgs/blob/main/autoware_sensing_msgs/msg/GnssInsOrientationStamped.msg) |  
| GNSS    |   /sensing/gnss/{gnss_name}/twist | geometry_msgs/TwistWithCovarianceStamped | 
| GNSS    |   /sensing/gnss/{gnss_name}/acceleration | geometry_msgs/AccelWithCovarianceStamped |  
| Vehicle  |  /vehicle/status/control_mode | autoware_auto_vehicle_msgs/ControlModeReport | 
| Vehicle | /vehicle/status/steering_status  |  autoware_auto_vehicle_msgs/SteeringReport | 
| Vehicle | /vehicle/status/velocity_status  |  autoware_auto_vehicle_msgs/VelocityReport |
| - | /tf |  | 
| - | /static_tf | 
| - | /map/pointcloud_map | sensor_msgs/PointCloud2
| - | /map/vector_map | autoware_auto_mapping_msgs/msg/HADMapBin | 

Some of the intermediate topics could include:
| topic name | message type | remark | 
  ---              |   ---                  |  --- |
| /localization/kinematic_state  | nav_msgs/Odometry | final position/velocity output |
| /localization/acceleration  | geometry_msgs/msg/AccelWithCovarianceStamped | final accelaration output |
| /localization/acceleration  | geometry_msgs/msg/AccelWithCovarianceStamped | final accelaration output |
| /perception/object_recognition/objects | autoware_auto_perception_msgs/msg/PredictedObjects | object recognition output | 
| /perception/traffic_light_recognition/traffic_signals | autoware_auto_perception_msgs/msg/TrafficSignalArray | traffic light recognition result | 
| /planning/mission_planning/route | autoware_auto_planning_msgs/msg/HADMapRoute | planned route | 
| /planning/scenario_planning/trajectory | autoware_auto_planning_msgs/msg/Trajectory | planned trajectory | 
| /control/command/control_cmd | autoware_auto_control_msgs/msg/AckermannControlCommand | control command output |