autowarefoundation/autoware-projects #27: Creating documentations for Autoware-compatible Simulators

sglee-morai, 2022-05-10:


sglee-morai, 2022-05-10:
### Description

To add information about simulators that support Autoware

(Draft) The document should include:
- a brief specifications
- installation guide
- quick start guide
- link for the full documentation

### Which Simulator?

1. TierIV Simulator (Default Simualtor)
2. MORAI SIM: Drive
3. CALRA* (Still needs a lot of development to make it fully compatible with Autoware)
4. LGSVL Simulator* (The fact that its support is stopped might be worth mentioning, just to prevent people from being confused)

WJaworskiRobotec, 2022-08-02:
First tutorial about running autoware.universe with AWSIM already merged to autoware-documentation  :  [link](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/digital-twin-simulation/awsim-tutorial/)

HuaweiAlgolux, 2022-08-02:
Is TierIV sim also Unity-based just like LGSVL?

sglee-morai, 2022-08-03:
Update from MORAI - @sglee-morai  will start working on this.

WJaworskiRobotec, 2022-08-03:
[link](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/digital-twin-simulation/awsim-tutorial/) to the AWSIM instruction

WJaworskiRobotec, 2022-08-03:
@HuaweiAlgolux yes, exactly. MORAI SIM as well.

HuaweiAlgolux, 2022-08-03:
> 

Wow. Thanks for letting me know. So both MORAI SIM and TierIV Sim are based on the Unity game engine. Great.

WJaworskiRobotec, 2022-11-09:
https://github.com/autowarefoundation/autoware-documentation/pull/232

xmfcx, 2025-05-21:
https://autowarefoundation.github.io/autoware-documentation/main/tutorials/

Goldob, 2025-05-21:
Closing as done.

closed 2025-05-21 by Goldob