autowarefoundation/autoware-projects #28: Creating a Bus Model (ITRI Bus)

sglee-morai, 2022-05-10:


PiotrRzRobotec, 2022-05-26:
The mesh and the texture for the ITRI bus model is ready. Now we are creating a collider and a unity package with HDRP materials.
[image]
[image]
[image]
[image]

PiotrRzRobotec, 2022-06-08:
The ITRI bus model is ready. However we are waiting for detailed information from ITRI company about dimensions and the rear lights. There are differences in dimensions in the documentation, we want to verify if ours are correct. Also, there is no information about the position of the rear lights (turn signal, break, reverse light). ITRI should give the answer this week.
[image]
[image]

sglee-morai, 2022-07-06:
# 22-07-06 Update from MORAI

Question regarding the bus model parameters
- What values/parameters are currently known? (whether specified by spec. sheet of the manufacturer or shared by ITRI)
- What values/parameters are to be shared by ITRI in the future?

I'm assuming the parameter list below would be identified by ITRI and be shared
- [planning simulator default values](https://github.com/autowarefoundation/autoware.universe/blob/main/simulator/simple_planning_simulator/param/simple_planning_simulator_default.param.yaml)
- [vehicle characteristics](https://github.com/autowarefoundation/autoware.universe/blob/main/simulator/simple_planning_simulator/param/vehicle_characteristics.param.yaml)

**I think we need to define the list of parameters & value for each param.**

WJaworskiRobotec, 2022-07-06:
3D Model is ready.