autowarefoundation/autoware-projects #32: Update launch files to enable Autoware Universe to run with Open AD Kit 1.0

LalithVipulananthan, 2022-05-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Verify that Autoware Universe can be run with Open AD Kit 1.0 on the same platforms (ADLINK AVA and Autocore PCU) and with the same CI/CD system. In short, we want to be able to apply the [existing Open AD Kit 1.0 Start Guide](https://gitlab.com/autowarefoundation/autoware_reference_design/-/blob/main/docs/Appendix/Open-AD-Kit-Start-Guide/index.md) to Autoware Universe instead of Autoware.Auto.

### Purpose

As a first step towards the next version of Open AD Kit, we need to confirm that Autoware Universe can be used with the existing Open AD Kit 1.0. In order to do, we need (new) launch files to run Autoware Universe as multiple instances of a single container image, then we need to test the two platforms that were originally supported by Open AD Kit 1.0

### Possible approaches

### Definition of done

- [ ] Provide all necessary launch files to enable Autoware Universe to run with Open AD Kit 1.0
- [ ] Verify that Autoware Universe can be run on the ADLINK AVA platform
- [ ] Verify that Autoware Universe can be run on the Autocore PCU platform
- [ ] Verify that Autoware Universe can be used with Tier IV's cloud-based CI/CD system

Sharrrrk, 2022-05-19:
@LalithVipulananthan I could also verify on Jetson AGX Xavier if required.

mitsudome-r, 2022-06-08:
Here's the list of the launch files that you can use:
https://github.com/autowarefoundation/autoware.universe/tree/main/launch

You can launch them with scenario simulator by following commands:
1. Download map+scenario from this link
2. Modify the path in downloaded scenario.yml to the path of downloaded map
3. Launch each modules:
* Vehicle Module:
```
```
*

ito-san, 2022-06-14:
@kasperornmeck

ito-san, 2022-06-16:
Successfully built with the latest master of meta-adlink-ampere and meta-ewaol.

ito-san, 2022-06-22:
Wating docker images for arm64 and scenario simulator from CI/CD team.

xmfcx, 2022-06-29:
> One is arm64 based image for Autoware

Autoware images are located here: https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe

Latest autoware arm build is in here: https://github.com/orgs/autowarefoundation/packages/container/autoware-universe/26844652?tag=galactic-latest-prebuilt-cuda-arm64

LalithVipulananthan, 2022-06-30:
Hi @Sharrrrk ! Have you been able to make any progress with your task of testing Autoware Universe with the Autocore PCU? Also, could you give me an idea when you will be able to test the Jetson AGX Xavier as well?

ito-san, 2022-07-02:
Now I'm getting an error regarding lanelet2. Investigating.
[image]

ito-san, 2022-07-02:
On my laptop. Having the same error but correctly shown. Seems to be another cause.
[image]
[image]

LalithVipulananthan, 2022-07-05:
@ito-san Thank you for your updates!

1. Were you able to get an x86/64 image for the Scenario Simulator?
2. If you load that lanelet2 map into Vector Map Builder, does VMB report the same errors/warnings about linestrings needing more than 3 different points?

ito-san, 2022-07-06:
@LalithVipulananthan 
> 1. Were you able to get an x86/64 image for the Scenario Simulator?

I intend to build from source.
[Installation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/scenario-simulation/planning-simulation/installation/)
And run.
[Scenario test simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/scenario-simulation/planning-simulation/scenario-test-simulation/)

> 2.  If you load that lanelet2 map into Vector Map Builder, does VMB report the same errors/warnings about linestrings needing more than 3 different points?

Looks like no problem in Vector Map Builder.
[image]

Next step, I will try to run Autoware with native environment, Ubuntu 20.04 on arm.

Sharrrrk, 2022-07-06:
> Hi @Sharrrrk ! Have you been able to make any progress with your task of testing Autoware Universe with the Autocore PCU? Also, could you give me an idea when you will be able to test the Jetson AGX Xavier as well?

@LalithVipulananthan I have just tested to launch all the modules on Autocore PCU and AGX Xavier, looks good so far. Next I will try to run all the modules with scenario simulator together.

ito-san, 2022-07-12:
Successfully run with planning_simulator on single container using old version of yocto packages
|package|used|latest|
|---|---|---|
|meta-adlink-ampere|v1.4|v2.1|
|meta-ewaol|v0.2.4|v1.0|

Here is what I've done.
On the AVA
```console
docker run --rm -it --net host -v /home/root/map:/map -v /home/root/cyclonedds:/etc/cyclonedds cbb36312a783 /bin/bash -c "export CYCLONEDDS_URI=file:///etc/cyclonedds/cyclonedds.xml; export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp; source install/setup.bash; ros2 launch autoware_launch planning_simulator.launch.xml map_path:=/map/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit launch_rviz:=false"
```
On the host
```console
rviz2 -d src/launcher/autoware_launch/autoware_launch/rviz/autoware.rviz
```

https://user-images.githubusercontent.com/57388357/178464934-7853cdee-b404-4f33-a1bd-d387d23f4232.mp4

LalithVipulananthan, 2022-07-19:
@kasperornmeck Would you be able to investigate why the Lanelet2 map did not display on AVA using the latest yocto builds, but did work with a previous version? If @ito-san can provide details of his testing and log files, that would be a good starting point, but failing that, do you have time to follow the same steps to see if you can repro the problem or not?

kasperornmeck, 2022-07-19:
@LalithVipulananthan 
Yes, looking through the documentation in this thread, I'd say it's sufficient for me to replicate. I would need is access to the container images (on AVA and on the host) and any data files.

ito-san, 2022-07-27:
@kasperornmeck I'm attaching console outputs of the old and new version of yocto packages. The outputs are when I simply launched autoware.

[ok_in_old_version.log](https://github.com/autowarefoundation/autoware-projects/files/9195901/ok_in_old_version.log)
[ng_in_new_version.log](https://github.com/autowarefoundation/autoware-projects/files/9195905/ng_in_new_version.log)

kasperornmeck, 2022-07-27:
@ito-san, I'm having issues accessing these files, could you send them to my email address instead? kasper.ornsteinmecklenburg[at]arm.com
Thanks!

mitsudome-r, 2022-07-28:
FYI, it seems like there was issues with Control module which caused autoware_state to never transition from Planning -> WaitForEngage for past few days. It should be fixed now with https://github.com/autowarefoundation/autoware.universe/pull/1450

kasperornmeck, 2022-07-28:
@mitsudome-r @ito-san 
Does that mean that we can rule out that the change in behavior wasn't due to the EWAOL version?

mitsudome-r, 2022-07-28:
@kasperornmeck I'm not sure what's the current failure in EWAOL, but if you are talking about [Lalith's comment](https://github.com/autowarefoundation/autoware-projects/issues/32#issuecomment-1189061103), at least the fix that I mentioned has nothing to do with lanelet2 map.

mitsudome-r, 2022-07-28:
@ito-san FYI, the newest docker image is pushed yesterday if you would like to try out the newest version of Autoware. https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/30491081?tag=galactic-20220728-prebuilt-cuda-arm64

ito-san, 2022-08-02:
@mitsudome-r @LalithVipulananthan 
Thank you for your kind support.
Made a little bit changes from OpenADKit 1.0. Confirmed that Autoware.universe runs as multiple k3 clusters and scenario_simulator_v2 works.
[image]
https://user-images.githubusercontent.com/57388357/182278607-5f1308ce-8706-4b7e-acf4-e44d72004947.mp4

kasperornmeck, 2022-08-02:
@ito-san 
This is really neat, well done! :)
Is there documentation for me to replicate this? I can do the OoB testing and give feedback on the documentation.

Sharrrrk, 2022-08-03:
My test report on PCU is here:
[Test report - Run Autoware.universe on PCU.pdf](https://github.com/autowarefoundation/autoware-projects/files/9247680/Test.report.-.Run.Autoware.universe.on.PCU.pdf)

Sharrrrk, 2022-08-03:
@LalithVipulananthan I've got a Jetson AGX Orin lately, I could do the test on it if requried.

LalithVipulananthan, 2022-08-03:
@Sharrrrk Hi Shark, if you can spare the time to test on the AGX Orin as well then that would be very much appreciated!

Sharrrrk, 2022-08-12:
Test report on Orin:
[Test report - Run AW.universe with OpenADKit v1.0 on Orin.pdf](https://github.com/autowarefoundation/autoware-projects/files/9315590/Test.report.-.Run.AW.universe.with.OpenADKit.v1.0.on.Orin.pdf)

Test report on Xavier:
[Test report - Run AW.universe with OpenADKit v1.0 on Xavier .pdf](https://github.com/autowarefoundation/autoware-projects/files/9315591/Test.report.-.Run.AW.universe.with.OpenADKit.v1.0.on.Xavier.pdf)

Sharrrrk, 2022-08-12:
@LalithVipulananthan  All the tests assigned to me are finised.

ito-san, 2022-08-14:
@LalithVipulananthan @kasperornmeck 
The pdf version of eSync OTA Update is available now.
[eSync OTA Update.pdf](https://drive.google.com/file/d/1hOvJChQthK3h-FWhEACmKdxoFV8dcD36/view?usp=sharing) (accessible only to the members who can access `AWF - Open AD Kit` Google Drive)

ito-san, 2022-08-17:
@kasperornmeck Confirmed that Autoware.universe runs on EWAOL 1.0 baremetal image.

https://user-images.githubusercontent.com/57388357/185083253-27234791-6f5e-47ce-baf8-454e0ca10140.mp4

ito-san, 2022-08-17:
@kasperornmeck Also confirmed arm64 version of `scenario_simulator_v2` works on AVA platform.

https://user-images.githubusercontent.com/57388357/185088263-c9dddcf6-b82d-45c6-ac6b-9437f50b7c72.mp4

kasperornmeck, 2022-08-17:
@ito-san very nice! :)

Does that mean that we can conclude that it is Xen (virtualization) which prevented Autoware.Universe from working properly on EWAOL v1.0?

ito-san, 2022-08-18:
@kasperornmeck Yes, maybe.

NilaySener, 2022-08-18:
@Sharrrrk Hi,  I was trying to run autoware modules with OpenAD kit, referring to the Orin test report you shared.The .pdf file you provided has incomplete commands due to the format of the file. 

e.g
[image]

 Can you share Orin's test report in a file format where I can access all the commands?

Sharrrrk, 2022-08-25:
@NilaySener you are right, I updated the report here:
[Test report - Run AW.universe with OpenADKit v1.0 on Orin.pdf](https://github.com/autowarefoundation/autoware-projects/files/9420635/Test.report.-.Run.AW.universe.with.OpenADKit.v1.0.on.Orin.pdf)
Please feel free to ask if you have any other concerns.

NilaySener, 2022-08-26:
@Sharrrrk Thank you for the report you shared. I followed the steps on Orin and got the same results.

Sharrrrk, 2022-09-06:
All tasks finished.

closed 2022-09-06 by Sharrrrk