autowarefoundation/autoware-projects #34: Document Autoware Universe ROS nodes

kasperornmeck, 2022-05-24:


LalithVipulananthan, 2022-05-31:
> FYI, here is the node configuration diagram of TIER IV's proposal created by @TakaHoribe. https://tier4.github.io/autoware-documentation/latest/design/node-diagram/#autoware-universe

@mitsudome-r Last week we confirmed that perception was the first component that we wanted to get information about, so is the diagram above an accurate representation of the ROS nodes required for perception?

mitsudome-r, 2022-06-01:
Here's the list of packages dependency of perception modules:
[image]

And here is the list of the packages in the graph:
```
autoware_auto_geometry_msgs
autoware_auto_mapping_msgs
autoware_auto_perception_msgs
autoware_auto_planning_msgs
autoware_auto_vehicle_msgs
autoware_cmake
autoware_lint_common
autoware_point_types
compare_map_segmentation
euclidean_cluster
grid_map_cmake_helpers
grid_map_core
grid_map_cv
grid_map_msgs
grid_map_pcl
grid_map_ros
ground_segmentation
image_projection_based_fusion
image_transport_decompressor
interpolation
kalman_filter
lanelet2_extension
lidar_apollo_instance_segmentation
map_based_prediction
multi_object_tracker
mussp
object_merger
object_range_splitter
occupancy_grid_map_outlier_filter
pointcloud_preprocessor
pointcloud_to_laserscan
shape_estimation
tensorrt_yolo
tier4_autoware_utils
tier4_debug_msgs
tier4_pcl_extensions
tier4_perception_launch
tier4_perception_msgs
traffic_light_classifier
traffic_light_map_based_detector
traffic_light_ssd_fine_detector
traffic_light_visualization
vehicle_info_util
```

kasperornmeck, 2022-06-03:
@mitsudome-r 
Is there some way to see which topics ROS nodes publish and subscribe to?

mitsudome-r, 2022-06-09:
You can just run the launch file and check it with rqt_graph.
e.g.,
`ros2 launch tier4_perception_launch perception.launch.xml mode:=lidar`
`ros2 run rqt_graph rqt_graph`

LalithVipulananthan, 2022-06-13:
@kasperornmeck Since the necessary commands have been provided, could you write up this information and submit a PR to the [autoware-documentation](https://github.com/autowarefoundation/autoware-documentation) repository?

The ["How to guides" section](
https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/) covers more advanced topics and so this would be the best place to write up a guide on how to list all the required packages and dependencies for a specific stack.

LalithVipulananthan, 2022-06-29:
PR created to add the information in this issue to Autoware documentation:
https://github.com/autowarefoundation/autoware-documentation/pull/162

closed 2022-06-29 by LalithVipulananthan