autowarefoundation/autoware-projects #35: Create an Open AD Kit documentation site

LalithVipulananthan, 2022-06-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Create a documentation site for Open AD Kit

### Purpose

At present, documentation for the Open AD Kit project is minimal, and the documentation that does exist is currently hosted on GitLab. As a foundation for better documentation going forwards, a documentation repository should be created and configured appropriately, before migrating all existing documentation to this site.

### Definition of done

- [x] Create an Open AD Kit documentation repository in the AWF org on GitHub
- [ ] Mirror the CI/CD configuration that is used by other Autoware documentation sites 
- [ ] Set up a GitHub pages site so that the documentation can be accessed
- [ ] Copy the Open AD Kit v1.0 quick start guide from GitLab to the new GitHub repository

kenji-miyake, 2022-06-08:
@LalithVipulananthan Should I help with something around these items?
> Mirror the CI/CD configuration that is used by other Autoware documentation sites
> Set up a GitHub pages site so that the documentation can be accessed

LalithVipulananthan, 2022-06-08:
@kenji-miyake Aha! I missed your comment and just asked you about this via Slack. If you could assist with those steps then that would be very much appreciated!

kenji-miyake, 2022-06-08:
@mitsudome-r @xmfcx @BonoloAWF Could someone approve to install `awf-autoware-bot` to https://github.com/autowarefoundation/open-ad-kit-docs? :pray: It seems I need approval from an org admin.
https://github.com/organizations/autowarefoundation/settings/apps/awf-autoware-bot

kenji-miyake, 2022-06-08:
@LalithVipulananthan I've changed the settings of GitHub Pages. You have to use `gh-pages` instead of `main` to show the site built by MkDocs.
[image]

LalithVipulananthan, 2022-09-13:
Closing as complete since the documentation repository was created and there is a public-facing website which has the documentation for Open AD Kit 1.0. 

Repository: https://github.com/autowarefoundation/open-ad-kit-docs
Website: https://autowarefoundation.github.io/open-ad-kit-docs/latest/

closed 2022-09-13 by LalithVipulananthan