autowarefoundation/autoware-projects #37: Create metrics to test the Perception-only container

LalithVipulananthan, 2022-06-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

### Purpose

In order to determine if the microservices PoC is feasible or not, we need to create some metrics that will allow us to make a meaningful comparison between the different potential ways of running Autoware:

- Built from source
- Single monolithic container
- Multiple monolithic containers, one for each Autoware component
- Microservice PoC architecture - running a Perception-only container with a container that has everything but Perception

### Possible approaches

### Definition of done

- [ ] List metrics and minimum/maximum acceptable values
- [ ] Record metrics for each of the possible architectures listed above
  - [ ] Built from source
  - [ ] Single monolithic container
  - [ ] Multiple monolithic containers, one for each Autoware component
  - [ ] Microservice PoC architecture - running a Perception-only container with a container that has everything but Perception
- [ ] Create a document that compares the recorded metrics

BonoloAWF, 2023-01-16:
Blocked by #36

kasperornmeck, 2023-01-20:
@BonoloAWF - I suggest we close this issue as it won't go anywhere and the issues this issue tries to solve is addressed in the DevOps Dojos.

closed 2023-01-22 by BonoloAWF