autowarefoundation/autoware-projects #40: Integration of ITRI campus with ITRI bus and Autoware - AWSIM

WJaworskiRobotec, 2022-07-19:


WJaworskiRobotec, 2022-08-02:
First integration of AWSIM with Autoware.universe in the ITRI campus environment : [link](https://drive.google.com/file/d/1tXTRF0II3q2s5MAszn-_rdLXOAj5i53m/view?usp=sharing)

WJaworskiRobotec, 2022-09-28:
Video showing integration of AWSIM with Autoware.universe in the ITRI campus and ITRI bus as Ego vehicle : [link](https://drive.google.com/file/d/1ZPP9R6zsvniRPuQVKEzIN8syMhAxTcJu/view?usp=sharing). Currently sample sensor kit is used as sensor configuration

WJaworskiRobotec, 2022-10-25:
I've managed to make the bus driving using cbus_vehicle_launch and itri_bus_sensor_kit with only one LIDAR. with 3 the non-existing obstacle is detected - looks like something is wrong with calibration of left and right sensor

[image]

WJaworskiRobotec, 2022-10-25:
Ok I have an AWSIM binary working with 3 OS1-64 lidars and using target vehicle model and sensor_kit

[image]

WJaworskiRobotec, 2022-11-07:
After running tests on the ITRI simulation environment, 2 issues were found : 

- https://github.com/autowarefoundation/autoware.universe/issues/2182
- https://github.com/autowarefoundation/autoware.universe/issues/2213

WJaworskiRobotec, 2022-11-23:
Videos presenting the progress with planning module improvements based on AWSIM can be found here : [link](https://drive.google.com/drive/u/1/folders/1lkJyVRQPTehjQR35RRBoeIaOwbNmnsng)

lchojnack, 2023-11-21:
As part of task https://github.com/autowarefoundation/autoware-projects/issues/70, I've prepared AWSIM simulation with ss2 support using ITRI environment (details: [link](https://github.com/autowarefoundation/autoware-projects/issues/70#issuecomment-1812813695))

During tests, I've encountered the following issue: EGO drives below ll2 according to Autoware, as presented in the video below. 

https://github.com/autowarefoundation/autoware-projects/assets/8416175/26ccf705-615a-425f-8754-8df2ad73b674

---
I've compared the ITRI environment with ll2 (from [here](https://evaluation.tier4.jp/evaluation/reports/846bee80-98bf-5529-aae9-a30892e833f8/tests/691dc5af-c0d3-5df4-b9b9-ae0539e1f88f/211b91d5-c6f0-52a3-a650-0b96ab743403/893bd875-82bf-5339-9ad6-80373bfb8fc5?project_id=awf)) in AWSIM. Some parts of ll2 are below the map (marked in orange) and some parts are above the map (marked in red)

[image]

[image]

---
I think that could be a good idea to remodel the ITRI map to fit the map with the ll2.

Goldob, 2025-05-21:
Details how to run the scene in AWSIM: https://github.com/autowarefoundation/autoware-projects/issues/70#issuecomment-1812813695

closed 2025-05-21 by Goldob