autowarefoundation/autoware-projects #44: Compatibility tests - sensors data publishers

WJaworskiRobotec, 2022-07-20:


SoohyeokPark-MORAI, 2022-08-03:
I am planning to start this Friday.
I made a sample configuration file that includes sensor-related publishers and publishers and subscribers just in case. 
[sample configuration file link](https://drive.google.com/file/d/1S4MESOJZuggkX5VmEUQl4Kt7ytw9xPyf/view?usp=sharing)

SoohyeokPark-MORAI, 2022-08-17:
camera, gps (gnss pose, pose with covariance), imu are under tests with MORAI SIM.

WJaworskiRobotec, 2022-09-07:
@SoohyeokPark-MORAI Is the PR for these tests ready ?

SoohyeokPark-MORAI, 2022-09-08:
@WJaworskiRobotec Here I share the link [PR link](https://github.com/autowarefoundation/autoware.universe/pull/1804)

WJaworskiRobotec, 2022-09-12:
@SoohyeokPark-MORAI thanks a lot, Have you tested it with AWSIM ?

SoohyeokPark-MORAI, 2022-09-12:
@WJaworskiRobotec I tried it with AWSIM but found an issue related to LIDAR whose data is not received.

WJaworskiRobotec, 2022-09-14:
OK I will check it with AWSIM with modified QoS

WJaworskiRobotec, 2022-09-21:
I have managed to make it work with AWSIM with some changes that I will push to the repo.

[image]

*4 failed tests for lidars are related to left and right sensors not included in the binary that I've tested

Hyeongseok-Jeon, 2022-10-25:
MORAI update
- I started to look the compatibility test code from @SoohyeokPark-MORAI 
- I could not specify the sensor configuration file yet

WJaworskiRobotec, 2022-11-23:
@Hyeongseok-Jeon any progress in this task ?

Hyeongseok-Jeon, 2022-11-23:
@WJaworskiRobotec There is no prograss on this task in terms of importing the sensor configuration file in compatibility test code. Sorry for delaying and we will work on this issue from next week (after finishing regular version update of our product)

Hyeongseok-Jeon, 2022-12-02:
@WJaworskiRobotec @SoohyeokPark-MORAI 

How about build configuration file with the sensor category and corresponding topics?
Sample for the config file will be below

```
{
  "Target_sim": [
    "AWSIM"
  ],
  "Camera_list": [
    "/sensing/camera/traffic_light/image_raw",
	"/sensing/camera/traffic_light/camera_info"
  ],
  "GPS_list": [
	 "/sensing/gnss/pose",
	 "/sensing/gnss/pose_with_covariance"
  ],
  "Lidar_list": [
    "/sensing/lidar/left/pointcloud_raw",
	"/sensing/lidar/left/pointcloud_raw_ex",
	"/sensing/lidar/right/pointcloud_raw",
	"/sensing/lidar/right/pointcloud_raw_ex",
	"/sensing/lidar/top/pointcloud_raw",
	"/sensing/lidar/top/pointcloud_raw_ex"
  ]
}
```

Based on this structure, compatibility test will be processed only for the topic in the above configuration file on the Target simulator.
Please let me know your opinion.

WJaworskiRobotec, 2022-12-12:
@Hyeongseok-Jeon 
In general looks fine, but I think we should add msg type to each topic, right ? 
For example  "/sensing/gnss/pose" and 
	 "/sensing/gnss/pose_with_covariance" have different message types, the same for image_raw and camera_info.

How about the structure like that:
```

{
    "Target_sim": [
      "AWSIM"
    ],
    "Camera_list": [
      {
        "topic":"/sensing/camera/traffic_light/image_raw",
        "msg": "..."
      },
      {
        "topic":"/sensing/camera/traffic_light/camera_info",
        "msg": "..."
      }
    ],
...
}
```

Hyeongseok-Jeon, 2022-12-20:
MORAI started deeloping the compatibility test tool which will be done before Jan 18th.

WJaworskiRobotec, 2023-01-16:
@SoohyeokPark-MORAI

Soohyeok-Park, 2023-01-26:
@WJaworskiRobotec @Hyeongseok-Jeon 
There have been some bug fixes and It's ready to be tested.
I have tested with the MORAI SIM.

WJaworskiRobotec, 2023-01-30:
@SoohyeokPark-MORAI 

Should I use humble or galactic ?

btw. Have you added some documentation of how to run the tests ? If not , it might be a good idea to add it a a separate page here : https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ under Digital Twin Simulations

Hyeongseok-Jeon, 2023-02-15:
@WJaworskiRobotec @Soohyeok-Park 

1. We test compatiblity testing tool based on humble
2. To run the test, please refer the followings
```
cd ~/${autoware installed directory}
source /opt/ros/humble/setup.bash
source install/setup.bash

cd ~/${autoware installed directory}/src/universe/autoware.universe/tools/simulator_test/simulator_compatibility_test/test_sim_common_manual_testing/
python3 -m pytest .
```
3. I will make PR for the compatilibty test manual soon.

Hyeongseok-Jeon, 2023-03-29:
@WJaworskiRobotec @SoohyeokPark-MORAI 

I created PR in the [link](https://github.com/autowarefoundation/autoware-documentation/pull/340)
Sorry for delaying and please let me know if you have any comment on the PR

Hyeongseok-Jeon, 2023-06-29:
@WJaworskiRobotec 

Does it have to be higher priority? Since Soohyeok is decided to leave MORAI, there will be some delay before hiring new developer.

WJaworskiRobotec, 2023-07-04:
Sure, no problem. It's definitely not the highest priority

Goldob, 2025-05-21:
Closing as stale.

closed 2025-05-21 by Goldob