autowarefoundation/autoware-projects #45: Design of experiments for BusODD High Fidelity Simulation

WJaworskiRobotec, 2022-08-02:


YoshinoriTsutake, 2022-08-03:
In my opinion, at first it would be better to know something like
- where Autoware is supposed to run in the map .
- how the map / route is 
  - how many lanes 
  - how long 
  - how large 
  - what road objects ( traffic lights, pedestrian crossing ) 
And it could be a test item to make Autoware move on the routes without NPCs  in a simulator to make sure mainly whether localization would work well or not. 

Then, additionally, to find other test items , it would be better to know 
- what kind of objects or NPCs are supposed to be in / around the route where Autoware moves 
- how NPCs are supposed to behave there 

And then, I think we would be able to start to discuss what scenarios could be tried out with AWSIM or MORAI SIM

WJaworskiRobotec, 2022-08-03:
[ODD WG scenarios](https://docs.google.com/presentation/d/1qiQMAT9Khnae4xWL36NUoMlNLZxSLn-HcMlEfg1BsBY/edit#slide=id.gd38c585fc3_0_0)

WJaworskiRobotec, 2022-08-04:
Components to be tested : 

**1. Localization**
- Launched only localization stack of Autoware: [localization launch](https://github.com/autowarefoundation/autoware.universe/tree/main/launch/tier4_localization_launch)
- Ground truth published by the simulator
- Vehicle controlled manually (or with some simple path following script) on the predefined paths
- Localization accuracy calculated using [localization evaluator](https://github.com/autowarefoundation/autoware.universe/tree/main/evaluator/localization_evaluator)

**2. Perception**
- Launched only perception stack of Autoware [perception launch](https://github.com/autowarefoundation/autoware.universe/tree/main/launch/tier4_perception_launch)
- Ground truth about visible objects published by the simulator (semantic segmentation to be added in the future)
- Ego vehicle is static
- Several scenarios with NPCs spawned around the ego vehicle - both static and dynamic.
- Perception algorithms accuracy calculated using perception evaluators (in progress)

**3. End-to-End testing**
- Launched entire Autoware stack
- Predefined paths + NPCs behavior
- Calculated metrics using all available evaluators

sglee-morai, 2022-08-08:
# Elements to Define a Test Scenario

### 1) Ego vehicle routes

### 2) NPC vehicles & pedestrians
- Random NPCs
  - Average No. of NPCs around the ego vehicle 
  - NPC behavior options
- Event-driven NPCs
   - Each NPC's route, behavior, etc.

### 3) Static Obstacles
- Which static objects are placed & where each one is placed (transform)

### 4) Time & Weather
- Time
- Weather Conditions

sglee-morai, 2022-08-08:
I'd like to give an example test scenario for localization node testing, which @WJaworskiRobotec mentioned in the first part of his comment. 

Please let me know whether this format is good enough or not, and of course let's talk about the contents of the scenario as well.

# Test Scenario Example for Localization Node Testing

### 1) Ego vehicle routes
(2 options in total)
1.1) ITRI Route 01
1.2) ITRI Route 02

ITRI Campus Route 01
[image]

ITRI Campus Route 02
[image]

### 2) NPC vehicles & pedestrians
(3 options in total)
- Random NPCs
  - Average No. of NPCs around the ego vehicle 
  - NPC behavior options
- Event-driven NPCs
   - Each NPC's route, behavior, etc.

2.1) No NPC. (Only the Ego vehicle is moving around)

2.2) Low Density
- Random NPCs
  - 5 vehicles around the ego vehicle on average. Including parked vehicles.
  - No pedestrians.
  - NPC behavior options (can be discussed later)
- Event-driven NPCs
  - No vehicle nor pedestrians

2.3) High Density
- Random NPCs
  - 10 vehicles around the ego vehicle on average. Including parked vehicles. 
  - No pedestrians.
  - NPC behavior options (don't care)
- Event-driven NPCs
  - No vehicle nor pedestrians

NOTE: how to define _around the ego vehicle_: can be discussed later)

### 3) Static Obstacles
3.1) No Static Obstacles

### 4) Time & Weather
4.1) Day & Sunny Weather

**Total Cases: 3 x 2 x 1 x 1 = 6**

WJaworskiRobotec, 2022-10-11:
**Components of AD stack to evaluate :**

- Perception (object detection)
- Localization
- Planning
- Control

To properly assess the quality of AD stack components, the predefined E2E scenarios needs to cover variety of road situations

**Road Situations:**

- Following another vehicle 
- Passing parked vehicles
- Meeting the vehicle driving in the other directions
- Intersection scenarios

[image]
[image]
[image]
[image]

To get the valuable output from the experiments, that will enable comparison of different versions of Autoware, the following metrics should be calculated and report from each scenario should be created.

**Metrics:**

- Overall result
  - T/F if vehicle managed to complete the scenario
- Planning/Control modules
  - Deviation metric
  - Obstacle distance
  - Time to collision 
  - Trajectory metrics
- Perception module:
  - Traffic light detection accuracy
  - Lidar segmentation
  - Lidar object detection / classification
  - Camera object detection / classification
- Localization module:
  - Localization accuracy metric

 

Due to the current limitations of AWSIM (No GT data for perception, no support for OpenSCENARIO), the following metrics can be calculated:

- Overall result
  - T/F if vehicle managed to complete the scenario
- Planning/Control modules
  - Deviation metric
  - Obstacle distance
  - Time to collision 
  - Trajectory metrics
- Localization module:
  - Localization accuracy metric

It’s not possible to create exact scenarios defining road situations. Because of that, the idea is to use Random Traffic feature to create scenarios covering some of the road situations. The most convenient way to do that would be to create one binary with controllable parameters of Random Traffic from the outside of the simulations (seed, number of NPCs) to play around with scenarios easily. Then we could create several configurations, and evaluate Autoware against a set of scenarios like that:

- Random Traffic Low 1 (5 vehicles)
- Random Traffic Low 2 (5 vehicles)
- Random Traffic Medium 1 (15 vehicles)
- Random Traffic Medium 2 (15 vehicles)
- Random Traffic High 1 (40 vehicles)
- Random Traffic High 2 (40 vehicles) 

All executed against both Path 1 and Path 2 of Ego Vehicle :

[image]
[image]