autowarefoundation/autoware-projects #46: Update documentation site to add OADK 2.0 installation instructions

LalithVipulananthan, 2022-09-13:
## Description

Now that the OADK [documentation repository](https://github.com/autowarefoundation/open-ad-kit-docs) and [website](https://autowarefoundation.github.io/open-ad-kit-docs/latest/) have been set up to explain setup and installation for the AVA Dev Platform and AutoCore PCU with Autoware.Auto, we need to add installation and setup instructions for OADK 2.0 with Autoware Universe running on those two platforms plus additional platforms (AGX Orin and AGX Xavier).

### Purpose

To maintain the documentation site with the most up-to-date information for Autoware Universe and tested/verified platforms, and to improve the visibility of the OADK and encourage greater usage.

### Definition of Done  

- [ ] Update the repository such that it is possible to have multiple versions of the OADK's installation and setup instructions
- [ ] Add setup instructions for Autoware Universe on the AVA Dev Platform 
- [ ] Add setup instructions for Autoware Universe on the AutoCore PCU
- [ ] Add setup instructions for Autoware Universe on the AGX Orin
- [ ] Add setup instructions for Autoware Universe on the AGX Xavier
- [ ] Add setup instructions to run multiple Autoware Universe containers orchestrated by K3s on the AVA Dev Platform

d-hashimoto, 2022-09-14:
> Update the repository such that it is possible to have multiple versions of the OADK's installation and setup instructions

After [this PR](https://github.com/autowarefoundation/open-ad-kit-docs/pull/27) is merged, it will be done.

evshary, 2022-09-26:
I've tried to sync the OADK 2.0 documentation site with branch proposal_v2.0 in [this PR](https://github.com/autowarefoundation/open-ad-kit-docs/pull/31).
Updating the tutorial to the latest version on the documentation site will help developers to find the tutorial much easier.
It'll be great if someone can help me to review the PR.

evshary, 2022-10-31:
> Add setup instructions for Autoware Universe on the AVA Dev Platform
I've verified the OADK 2.0 tutorial on the documentation site, and I'm able to install & run scenario simulator successfully.

BonoloAWF, 2023-01-18:
@evshary can this Issue be closed given that https://github.com/autowarefoundation/open-ad-kit-docs/pull/31 is merged?

evshary, 2023-01-18:
@BonoloAWF For AVA platform, I think the answer is yes.
And I found that Autocore has also put Xavier platform into the documentation
https://github.com/autowarefoundation/open-ad-kit-docs/pull/38
Therefore, in my opinion, maybe this issue can be closed now, but I'm not sure whether there are any concerns from others.

closed 2023-01-19 by BonoloAWF