autowarefoundation/autoware-projects #47: Createing a Simulation Environment (ISUZU project)

WJaworskiRobotec, 2022-09-28:


MertClk, 2022-09-30:
Point cloud map and Lanelet2 map of our test campus can be found below.

PCD:  [Map link](https://drive.google.com/file/d/1O14cNaT3WlPM4qbLADotCteKqRk236Pi/view?usp=sharing)
HAD: [Map link](https://drive.google.com/file/d/1anarT0IW_8Ou2BdRHjtc--dmr-vWUCAP/view?usp=sharing)

Hyeongseok-Jeon, 2022-10-11:
MORAI are internally processing the 3D map construction.
But constructing full 3D map will take a long time.

We need to discuss on this issue.

[image]

WJaworskiRobotec, 2022-10-25:
@Hyeongseok-Jeon when we can expect the model with simplified buildings ?

Hyeongseok-Jeon, 2022-11-02:
Several issues in creating 3D env for ISUZU project

1. width of the road is too tight
- issue on the bidirectional road
[image]

- issue on the intersection -> when using our internal NPC algorithm, vehicle cannot trun left within the road mesh
[image]

2. issues on the hd map data (lanelet2) for z axis value
[image]

3. traffic signal issue
- based on the data, there is a traffic sign which can not be seen from the Google road view
[image]

- based on the data, every traffic sign is places over 10m compared to the road

If the above issues don't need to be considered, we will try to show the map with satelite image on next meeting (without building models). After selecting the buildings to be modeled from the satelite image, we can announce the due date for providing the map asset with simplified buildings.

mehmetdogru, 2022-11-04:
@Hyeongseok-Jeon 

Thank you so much for working on this.

> * issue on the bidirectional road

[image]

If that is the road shown above, we had to annotate that road overlapping with each other it is simply because that road needs to be bidirectional for routing purposes yet there is not enough space for two separate (not overlapping) roads. Since there is no other way, we can leave this one as it is.

> * issue on the intersection -> when using our internal NPC algorithm, vehicle cannot trun left within the road mesh

It is not clear to me where is this intersection or where the NPC tries to turn to but we do have problems on the turns especially for the long vehicles. So we are extending the lanelet borders in autoware.universe for long vehicles to be able to complete the turns which originally too narrow for them. I am guessing your internal NPC's are not using autoware to navigate so I am afraid this is a normal behavior if the NPC is a long vehicle. We don't want to tailor lanelet2 map much for the bus giving the HAD-Map annotation shouldn't be done wrt a specific vehicle.

>  issues on the hd map data (lanelet2) for z axis value

May I ask if these defects are common or on just some spesicif places?

> * based on the data, there is a traffic sign which can not be seen from the Google road view

We are using a dummy traffic light (a portable one). There is not a real traffic light on the annotated spot.

Hyeongseok-Jeon, 2022-11-05:
>May I ask if these defects are common or on just some spesicif places?
Currently, we found that this issue is only for the specific place (node id : 175). If you found another nodes with data defection, we will let you know.

mehmetdogru, 2022-11-06:
> Currently, we found that this issue is only for the specific place (node id : 175). If you found another nodes with data defection, we will let you know.

@mozhoku Could you also look at this issue if we have it on more places?

> Can you specify the exact 3 dimensional position and shape of the traffic light? Can we use the traffic light model provided by armağan via discord as below?

@Hyeongseok-Jeon 
As far as I know the position of traffic light (position of bulbs) as it is specified in lanelet2 map with `light_bulbs` tag. @mozhoku can you verify this information?

`<tag k='type' v='light_bulbs' />`

Hyeongseok-Jeon, 2022-11-08:
We attached satellite image on the road mesh but 3d modeling for the building is on-going and the building to be modeled by MORAI should be discussed.

[image]

WJaworskiRobotec, 2022-11-09:
@mehmetdogru Please provide the information about the buildings. Video, estimation of height, priority of the modeling

mehmetdogru, 2022-11-09:
@WJaworskiRobotec @Hyeongseok-Jeon 

Tomorrow @senagolcuk and some other of my colleagues will be on the campus. They will provide you with some video recordings related to the buildings around.

@xmfcx will provide you with the list of buildings which should be modeled first and a dense pcd map to get the height of the buildings properly.

And we will share the test route asap.

senagolcuk, 2022-11-09:
> @WJaworskiRobotec @Hyeongseok-Jeon
>
> Tomorrow @senagolcuk and some other of my colleagues will be on the campus. They will provide you with some video recordings related to the buildings around. [...]

@WJaworskiRobotec @Hyeongseok-Jeon @mehmetdogru 

I am sharing the test route below. I will share the videos of the campus as soon as possible.

[image]

xmfcx, 2022-11-09:
Isuzu environment dense point cloud map LAS file (3.7 GB): https://drive.google.com/file/d/14Fn9gBC7UrSyr7aytPKL7NqHZeWTD3_Z/view?usp=share_link

We'll share the list of buildings and a video recording of the route too.

mozhoku, 2022-11-09:
> > Currently, we found that this issue is only for the specific place (node id : 175). If you found another nodes with data defection, we will let you know.
> 
> @mozhoku Could you also look at this issue if we have it on more places?

There are no other issues as big as that one in node heights. Sorry, I must have overlooked that one. 🙏 I've updated the map and can share with @Hyeongseok-Jeon again if it's necessary. _(Our project manager is on vacation so I don't know if I can share the map publicly. I can send it from Discord if needed.)_

> As far as I know the position of traffic light (position of bulbs) as it is specified in lanelet2 map with `light_bulbs` tag. @mozhoku can you verify this information?
> 
> `<tag k='type' v='light_bulbs' />`

Yes, you can use the light bulb positions in HDmap. It should be fine. 👍

WJaworskiRobotec, 2022-11-09:
@xmfcx @senagolcuk @mehmetdogru Thanks a lot for really quick delivery of missing pieces !

xmfcx, 2022-11-15:
@WJaworskiRobotec here is a recording of the environment and the map: 

[image]

The video is a bit choppy due to screen recording and visualizing the cameras without optimization. But it should give an idea about how campus looks in general.

xmfcx, 2022-11-15:
@WJaworskiRobotec I've also marked the prominent buildings on the map (not particularly sorted by importance)

[Google Earth Link](https://earth.google.com/web/@40.80859577,29.36148672,9.06036614a,2934.92322443d,30.00059455y,0h,0t,0r/data=MigKJgokCiAxQ0dzcF9ESDhVbmhTU1BQcU9xZVVfbEFpVFNMeGpGQSAC)

[Google Maps Link](https://www.google.com/maps/d/u/0/edit?mid=1CGsp_DH8UnhSSPPqOqeU_lAiTSLxjFA&usp=sharing)

Should look like this:

[image]

WJaworskiRobotec, 2022-11-23:
@Hyeongseok-Jeon what is the progress of this task ?

Hyeongseok-Jeon, 2022-11-23:
@WJaworskiRobotec

Currently we calculated the height of the aforementioned buildings based on LAS file and creation process is on going

WJaworskiRobotec, 2022-11-23:
Lets test in the ss2 with LL2 and vehicle model.

Hyeongseok-Jeon, 2022-11-25:
We finished creating 3D env for ISUZU project with simplified buildings. It will be uploaded to drive shortly.

[image]
[image]
[image]
[image]

xmfcx, 2022-12-05:
@Hyeongseok-Jeon

The level of detail in the 3D map is impressive. It's clear that you put a lot of time and effort into creating it. This high-quality map that will be very useful for our simulations.

I love the way you've incorporated different elements into the map, such as roads and buildings. Great job!

We will start testing the bus in this map and provide feedback.

Thank you for your hard work on this project.

mehmetdogru, 2022-12-07:
@Hyeongseok-Jeon 

 Are this environment and ISUZU bus ready to try on the same scene in AWSIM?

mehmetdogru, 2022-12-07:
Necessary component are shared via G-Drive. @mozhoku from LeoDrive will give it a try to integrate these component into AWSIM.

mehmetdogru, 2022-12-21:
@Hyeongseok-Jeon 

@djargot has reported some problems [here](https://github.com/autowarefoundation/autoware-projects/issues/51#issuecomment-1361198178) related to the environment during he integrates it into AWSIM. Is it possible for you to check it out?

mehmetdogru, 2022-12-30:
[Gebze Technical University map](https://github.com/autowarefoundation/autoware-projects/issues/47#issuecomment-1263462750) origin conf:

```yaml
lanelet2_map_projector_type: UTM   # Options: MGRS, UTM
latitude: 40.81187906        # Latitude of map_origin, using in UTM
longitude: 29.35810110          # Longitude of map_origin, using in UTM
```

Please replace this conf values here to be able to use the map with autoware.universe:

https://github.com/autowarefoundation/autoware.universe/blob/b844cb12e9bf7e835f7c93a5a0c2233b15543de3/map/map_loader/config/lanelet2_map_loader.param.yaml#L3-L5

WJaworskiRobotec, 2023-01-18:
3D Environment was succesfully tested in AWSIM. Issue can be closed.