autowarefoundation/autoware-projects #48: Creating a ISUZU bus model

WJaworskiRobotec, 2022-09-28:


MertClk, 2022-09-30:
We have a blend file for the bus. You can find it [here](https://drive.google.com/file/d/1LCyJux9zM4V7xSMID5ETAzDsxjJ8T7_e/view?usp=sharing)

WJaworskiRobotec, 2022-10-12:
Vehicle files, including vehicle launch and sensor_kit launch : [link](https://drive.google.com/drive/folders/1bkxamrkKsbP3J5339GVf5yKhJL0iii4l)

WJaworskiRobotec, 2022-10-12:
I have integrated ISUZU bus with AWSIM, together with sensors as descirbed in the documentation. Currently working on running it with Autoware

[image]

Hyeongseok-Jeon, 2022-10-25:
MORAI is ready for the ISUZU bus model but I need to check later since i am currently on the business trip.

Hyeongseok-Jeon, 2022-11-02:
We created the ISUZU bus model in MORAI SIM and dynamic setting is on going which will be completed soon.

[image]
[image]

WJaworskiRobotec, 2022-11-03:
@Hyeongseok-Jeon 
The model looks really great ! Could you share the .unitypackage with the vehicle ?

Hyeongseok-Jeon, 2022-11-14:
.unitypackage file for ISUZU bus model is uploaded. Be aware that the copyright (including the MORAI logo) MUST NOT be removed in further usage.

WJaworskiRobotec, 2022-11-23:
It will be tested/reviewed by TierIV