autowarefoundation/autoware-projects #51: Integrate GTU Campus Environment with ISUZU Bus Model into AWSIM

mehmetdogru, 2022-12-07:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

It is aimed that under this issue, results/findings/problems about integration of GTU (Gebze Technical University) Campus environment with ISUZU bus model can be shared and discussed.

Related issues: 

**GTU environment:** https://github.com/autowarefoundation/autoware-projects/issues/47
**ISUZU bus model:** https://github.com/autowarefoundation/autoware-projects/issues/48

FYI: @WJaworskiRobotec

### Purpose

To test GTU campus with ISUZU bus in simulation.

### Possible approaches

Integrate the components provided via G-Drive.

### Definition of done

- [ ] Import unitypackages provided and build the project.
- [ ] Test the environment & bus model

mozhoku, 2022-12-07:
I will work on this issue. 🤗

djargot, 2022-12-21:
The map asset is NOT created using High Definition Render Pipeline (HDRP), so when importing the unitypackage to TierIV SIM the materials attached in the file need to be converted to HDRP. It would be great if you could share the unitypackage in HDRP. 
Also, material for road looks incorrect:
[image]
Firstly, all parts of road have the same material (crossings, road, lines), secondly the material does not have normals (see the warning in the picture above), so reflections cannot be computed.

Hyeongseok-Jeon, 2022-12-21:
@djargot @WJaworskiRobotec @mehmetdogru 

Thank you for your information. I will try to update unitypackage file within this month.
Can you specify the version of the HDRP as well?

djargot, 2022-12-22:
@Hyeongseok-Jeon 
Thank you for the message! We are using HDRP 11.0.0.

djargot, 2022-12-23:
@mehmetdogru 
I have tried to use ISUZU files provided here: https://drive.google.com/drive/u/1/folders/1bkxamrkKsbP3J5339GVf5yKhJL0iii4l
but there is a problem with running Autoware on Ubuntu 22 with ROS humble, as I could not find Applanix driver package for ROS humble. Is there such a package available?

mehmetdogru, 2022-12-23:
@djargot 

Hi. Thank you for working on this. 

You can clone it from [here](https://github.com/vautonomous/applanix.git). `humble_on_vehicle` branch should work fine afaik.

If you will have any further issues related to this, please let us know. @kaancolak or @meliketanrikulu can give you more detailed answers.

Hyeongseok-Jeon, 2022-12-26:
@djargot I uploaded the modified version of the ISUZU simulation environment Please check

djargot, 2023-01-10:
Regarding vehicle launch and sensor kit (https://drive.google.com/drive/folders/1bkxamrkKsbP3J5339GVf5yKhJL0iii4l) I have several comments that I would like to discuss.
The values for the sensor kit calibration seem to be a little off:
sensor_kit_base_link and lidars:
```base_link:
  sensor_kit_base_link:
    x: 6.20
    y: -0.125
    z: 0.17
    roll: -0.037
    pitch: 0.20070000000000002
    yaw: -0.03
```
The coordinate system diverges from vertical axis of the vehicle and thus, as can be seen in the picture below, lidars are translated to the one side of the bus.
[image]

GNSS:
The gnss coordinate system is below the ground level. Even though the coordinate system placement is not crucial for the correct localization it was probably not designed to be there? 
[image]

Camera:
Top camera placement is inside the bus which diverges for the diagram provided. I am not sure if the diagram reflects reality, but it makes more sense the way it is shown on the diagram compared to what I obtained from calibration values.
[image]
[image]

Final remark (not crucial for the development but worth to be noted), is that the model for the bus was not obtained using HDRP (I manually converted shaders to HDRP to be able to use the model with AWSIM).

djargot, 2023-01-13:
So, I modified the sensor kit and I was able to integrate AWSIM (https://github.com/tier4/AWSIM/tree/humble) to support Isuzu project. I have used two different sensors configurations: one with Isuzu bus equipped with 3 lidars and 1  camera, and the second one with the bus equipped with 3 lidars and 3 cameras. The former is enough to make it run with Autoware main branch. To see the teaser visit: https://drive.google.com/file/d/1cE6NGYR0VMCqcPGtX5z4x-lQ6MntQqJn/view?usp=share_link
The video was prepared using AWSIM humble branch and Autoware main branch.
As TIER IV is currently working on updating the main branch of AWSIM to support ROS humble we will share the binaries to the public very soon.

mehmetdogru, 2023-01-15:
@djargot 

> So, I modified the sensor kit and I was able to integrate AWSIM

Thank you for the work, it looks really nice :100: 

> Unfortunately, there is a problem that we encountered, namely that speed bumps are detected as obstacles by Autoware.

That is an interesting one :smile: It is hard for me to say from this picture but only way should be (please correct me if I am wrong) that in AWSIM, speed bumps are modeled as 3D meshes. Evidently, as the result of `ground_segmentation` with the current default configuration speed bumps perceived as obstacles on the road. 

We don't use `ground_segmentation` package with the default configuration. So I can say that we don't have this issue while we are testing the bus on the campus. Also we have some minor modifications and parameter tuning on some repositories. Maybe for tests with AWSIM we can use[ forked repository of autoware for isuzu project](https://github.com/vautonomous/autoware) since this is the repository we use on the field. 

**PS:**  If you choose to preceed with isuzu repository please make sure that:
- You check out to `on-vehicle` [branch](https://github.com/vautonomous/autoware/blob/on-vehicle/autoware.repos) on `autoware` repository.
- You install ArenaSDK as instructed on[ README file](https://github.com/autowarefoundation/lucid_vision_driver#readme), otherwise `lucid_vision_driver` package won't build. As a second option you can comment out [the driver](https://github.com/vautonomous/autoware/blob/248ca8def59e6fdeb43547fad2bb9cda084e503c/autoware.repos#L74-L77) and it wouldn't be pulled since you won't be needing it anyways in simulation.
- Proceed the installation [as instructed](https://autowarefoundation.github.io/autoware-documentation/main/installation/).

We can't wait to get sim binary to test it on our end. Thanks again.

closed 2023-03-16 by WJaworskiRobotec