autowarefoundation/autoware-projects #52: Verify that Autoware Universe can be run with Open AD Kit 2.0

cyn-liu, 2022-12-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Verify that Autoware Universe can be run with Open AD Kit 2.0 on the same platforms (Autocore PCU and Jetson AGX Orin and Jetson Xavier) .

### Purpose

Verify that Autoware Universe can be run with Open AD Kit 2.0 on the same platforms (Autocore PCU and Jetson AGX Orin and Jetson Xavier) .

### Possible approaches

All test are finished.

### Definition of done

My test report on PCU/Orin/Xavier is here:

[Test report - Run autoware.universe.pdf](https://github.com/autowarefoundation/autoware-projects/files/10184800/Test.report.-.Run.autoware.universe.pdf)

cyn-liu, 2022-12-19:
We use `K3s` and `Podman`  to run Autoware respectively, and compare the time of starting Autoware with the two methods.
This is our test report:
[Test report - Run AW.universe with k3s and podman.pdf](https://github.com/autowarefoundation/autoware-projects/files/10257701/Test.report.-.Run.AW.universe.with.k3s.and.podman.pdf)

liuXinGangChina, 2022-12-26:
Hi, Fatih @xmfcx .
Usually whom should we assign to review this kind of issue?

thanks!