autowarefoundation/autoware-projects #54: Select and run demo scenarios for Bus ODD demo

WJaworskiRobotec, 2023-01-31:


WJaworskiRobotec, 2023-01-31:
1. Moving in Harmony with a Lead Vehicle

● Bus follows up with a lead vehicle at around 30km/h

● Bus slows down if the lead vehicle is slowing down

2. A Vehicle Cuts-In with the Bus

● Bus drives on the right lane

● Other vehicle drives on the left lane, slightly ahead of the bus

● Other vehicle cuts in front of the bus (from the left lane to the right lane)

● Bus slows down and allows another vehicle to get into the lane safely

3. Performing Lane Change to Pass a Stationary Vehicle

● Bus drives on the right lane

● Bus detects a stationary vehicle on the path

● Bus slows down and safely manoeuvres around the stationary vehicle by switching lane

● Bus merges back to the right lane after passing the stationary vehicle at a safe distance

4. Interacting with Pedestrians at a Crosswalk

● Bus drives on the route

● Bus detects some pedestrians standing by the crosswalk

● Bus predicts that pedestrian might cross the road and slows down

● Bus slows down gently until coming to a complete stop before the crosswalk and waits until pedestrians cross

● Bus continues to drive after the crosswalk passing

5. Interacting with a Speed Bump

● Bus drives on the route

● Bus detects a speed bump coming up

● Bus slows down gently until it arrives at the speed bump

● Bus gently passes over the speed bump and continues driving as normal

6. Interacting with a Traffic Light

● Bus drives on the route

● Bus detects a traffic light coming up

● Bus detects the colour of the traffic light and

○ Comes to a stop if the traffic light is RED

○ Bus continues on driving if the traffic light is GREEN

○ Bus continues on driving (?) if the traffic light is YELLOW

7. T-Junction Turn and Interacting with Incoming Traffic

● Bus drives towards the T-Junction

● Bus comes to a complete stop before the T-Junction line

● Bus stops until the passing vehicle drives through at the main road

● When the passing vehicle is at a safe distance, the bus merges to the main road with a right turn

8a. Bus Stop Handling (onboarding passengers)

● Bus drives towards the upcoming bus stop

● Bus slows down while reaching the bus stop

● Bus docks at the bus stop

● Bus opens the doors and allows passengers to get on the bus

● Bus closes the doors after all passengers are onboarded

● Bus drives out of the bus stop area

8b. Bus Stop Handling (offboarding passengers)

● Bus drives towards the upcoming bus stop

● Bus slows down while reaching the bus stop

● Bus docks at the bus stop

● Bus opens the doors and allows passengers to get off the bus

● Bus closes the doors after all passengers are offboarded

● Bus drives out of the bus stop area

YoshinoriTsutake, 2023-02-03:
@WJaworskiRobotec 
From my point of view, the scenario " 1 " seems fine. 
On the other hand, as for the scenario " 3 " , could you let me make sure how many lanes for going forward are ? 
If only one , and if you would expect Ego to switch to the lane for the opposite direction, we have to make sure whether  Autoware has the feature. 
Actually I heard from a coworker that the feature has been under development.

WJaworskiRobotec, 2023-02-03:
We have a 2-lanes road for scenario 3 in GTU campus - demo site.