autowarefoundation/autoware-projects #56: To update a README

WJaworskiRobotec, 2023-04-12:


Hyeongseok-Jeon, 2023-05-15:
@WJaworskiRobotec 
As I shared via DM, this is the first draft for the Sim WG's goal. 
I think it will be better to define the WG's goal before move to the current activities

# Working group goals

The goals of the Simulation Working Group are:

1. Define the minimum requirement for each E2E simulation for being compatible with Autoware
1. Propose simulation based DevOps pipeline and provide use cases for developing each functional elements within the pipeline
1. Provide various opportunities for the Autoware developers to use the E2E simulators
1. Secure the scalability in terms of map and vehicle assets within the E2E simulators for the various use case of Autoware
1. Co-work with other Working Groups by supporting virtual/synthetic environment

Goldob, 2025-07-02:
I updated the wiki page by
- updating the "Working group leaders" section
- removing the "Current activites" section due to containing outdated information
- replacing the above with a new "Roadmap" section

@xmfcx please take a look if that seems OK to you.