autowarefoundation/autoware-projects #63: Creation of 3D model of the highway environment

WJaworskiRobotec, 2023-04-25:
Simulation environment enabling testing of highway use cases for 
passenger cars. The following deliverables should be provided:
- 3D environment model
- Lanelet2 vector map
- 3D environment integrated in AWSIM

WJaworskiRobotec, 2023-04-26:
Tool for LL2 creation : https://tools.tier4.jp/vector_map_builder_ll2/

Hyeongseok-Jeon, 2023-05-16:
Two virtual highway environment based on Unity (not HDPR) engine is implemented in the MORAI SIM
We will prepare HDPR version of unitypackage file for the environment including the HD map data with LL2 format

[image]
[image]

WJaworskiRobotec, 2023-05-17:
In parallel, Robotec is working on another highway environment

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

WJaworskiRobotec, 2023-05-23:
Update of the Robotec highway environment

[image]
[image]

Hyeongseok-Jeon, 2023-05-31:
@WJaworskiRobotec (cc: @Igata-ctrl )

We finished constructing a 3D env including LL2 map file.
Is there any link to share the files (unitypackage, osm)?

Hyeongseok-Jeon, 2023-06-04:
@Igata-ctrl (cc: @WJaworskiRobotec )

I uploaded the materials in the ODD WG drive
https://drive.google.com/drive/folders/1nDZUTut8_RYXaewk0NpPH0DYsGHlb7ZR

Please let me know if the directory is ready for the SIM WG

WJaworskiRobotec, 2023-06-07:
We have finished our alternative highway model. Now started working on the LL2.

[image]
[image]
[image]
[image]
[image]
[image]

Hyeongseok-Jeon, 2023-06-14:
.pcd file is added to https://drive.google.com/drive/folders/1nDZUTut8_RYXaewk0NpPH0DYsGHlb7ZR

Hyeongseok-Jeon, 2023-06-29:
@WJaworskiRobotec @Igata-ctrl 

Another virtual highway environment from MORAI is uploaded.

Igata-ctrl, 2023-07-05:
@Hyeongseok-Jeon (@WJaworskiRobotec) 
Thank you for adding another virtual highway map! I took my liberty to upload it to Autoware Evaluator (CI/CD pipeline) and I actually prototyped a couple of simple scenarios on both maps.
Unfortunately, it seems these test scenarios fail, but Mitsudome-san is kindly looking into it to identify the cause.
Thanks!

Goldob, 2025-05-21:
Keeping the issue for now. Requirements will be discussed with the PoV WG.

closed 2025-11-05 by Goldob