autowarefoundation/autoware-projects #66: To verify Highway 3D model with Autoware

WJaworskiRobotec, 2023-07-06:
### Checklist
- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description
3D environment : https://drive.google.com/drive/u/1/folders/1nDZUTut8_RYXaewk0NpPH0DYsGHlb7ZR

### Purpose
- To check the highway environment provided by the Simulation WG
- To verify how Autoware's localization stack will work in the environment without characteristic features

### Possible approaches
-

### Definition of done
- Import 3D environment to AWSIM
- Run Autoware
- provide a report/video with results/problems

lchojnack, 2023-07-17:
I've run Autoware with provided 3D environments of virtual highway 1 and virtual highway 2. The provided ll2 and PCD do not match each other (checked in Vector Map Builder) and for that reason, I've generated the new PCD using [PointCloudMapper](https://tier4.github.io/AWSIM/Components/Environment/PointCloudMapper/). 

To Google Drive, I've added ll2, PCD, and AWSIM binary which I've used for testing. 

# AWSIM scene preparation

I've had to set the MGRS grid zone and offset position for both environments. I've tried to adjust the MGRS offset position as precisely as I cloud to do it. I've set the MGRS zone for both scenes to `11LST` and the MGRS offset position to:
- virtual highway 1 -> `X: 64598.66`, `Y: 80450.12`, `Z: -383.9825`
- virtual highway 2 -> `X: 64613.00`, `Y: 80443.00`, `Z: 0`

As the EGO vehicle I've used the Lexus prefab. 

As mentioned earlier, I've generated the new PCD for the scenes. 
In order to load ll2 to AWSIM (needed for PontCloudMapper), I've had to open ll2 in VMB and then save it.

# Autoware

I've used the following command
```
ros2 launch autoware_launch e2e_simulator.launch.xml vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit map_path:=$HOME/autoware_map/virtual_highway_1/
```

# Results
## Virtual Highway 1

### Video
https://github.com/autowarefoundation/autoware-projects/assets/8416175/f6d66448-48df-462b-879a-7dd289a4df7e

### Comment
In this scene, the EGO vehicle is not able to follow the planned trajectory, and as a result the vehicle crashes. It seems to be the issue with the localization. 

## Virtual Highway 2

### Video
https://github.com/autowarefoundation/autoware-projects/assets/8416175/1668327b-6d08-4465-9898-0a2567ca4361

### Comment

The EGO is able to follow the planned trajectory and achieve the goal position. The localization seems to working correctly.

mitsudome-r, 2023-07-17:
Thanks for sharing the highway maps.
I have downloaded virtual highway 2 map, but it seems like the lanes in red box have solid lines are tagged as non-lane-changeable lanes. Is this intended?

[image]

Hyeongseok-Jeon, 2023-07-26:
@mitsudome-r 

I uploaded new LL2 map to google drive as "Virtual_highway_morai_2_mod.osm". 
Please check

Goldob, 2025-05-21:
Keeping the issue for now. Requirements will be discussed with the PoV WG.

Goldob, 2025-11-05:
Closing as superseded by #74