autowarefoundation/autoware-projects #67: AWSIM feedback

WJaworskiRobotec, 2023-08-17:


xmfcx, 2023-08-22:
@TakaHoribe has tested it and made a post in here:
- https://github.com/orgs/autowarefoundation/discussions/3741#discussioncomment-6779568

I had made 3 tests.

## System & test setup
CPU: 5900X (12c24t)
GPU: RTX 3090 (24GB VRAM)
Memory: 64GB

I've used the commits:
- https://github.com/tier4/AWSIM/commit/d7d1d42a379b1996f4d00f21850e46074647fe21
   - Added https://assetstore.unity.com/packages/tools/gui/graphy-ultimate-fps-counter-stats-monitor-debugger-105778 to show the performance graphs
   - Enabled side lidars to have better perception performance (it worked with the launch file on autoware side without modifications)
   - Compiled a build because running from editor has worse performance.
- And latest autoware from with these frozen hashes:
<details>
  <summary>autoware-sim-test.repos</summary>

```yaml
repositories:
  core/autoware.core:
    type: git
    url: https://github.com/autowarefoundation/autoware.core.git
    version: 99891401473b5740e640f5a0cc0412c0984b8e0b
  core/autoware_adapi_msgs:
    type: git
    url: https://github.com/autowarefoundation/autoware_adapi_msgs.git
    version: fcdd3d07814b91090cd60079bd378396da6924de
  core/autoware_common:
    type: git
    url: https://github.com/autowarefoundation/autoware_common.git
    version: 58039c8b598d7e247994015d9c10c5b39e36b545
  core/autoware_msgs:
    type: git
    url: https://github.com/autowarefoundation/autoware_msgs.git
    version: 62f341d97515c81236a9a6700f02ca4e8df02304
  core/external/autoware_auto_msgs:
    type: git
    url: https://github.com/tier4/autoware_auto_msgs.git
    version: e1795354161ed54d66659366262578863bd2b862
  launcher/autoware_launch:
    type: git
    url: https://github.com/autowarefoundation/autoware_launch.git
    version: f5cd323f75dd8e5c9a210098970aa082b1be1a80
  middleware/external/heaphook:
    type: git
    url: https://github.com/tier4/heaphook.git
    version: ec260d7a7f06f29c04de2c846fa6d2158e770877
  param/autoware_individual_params:
    type: git
    url: https://github.com/autowarefoundation/autoware_individual_params.git
    version: 9b8b4064bf856dbd3f781a5ebd1f96135496878c
  sensor_component/external/nebula:
    type: git
    url: https://github.com/tier4/nebula.git
    version: 189d7e273aa66a528becac30024db5c96fa645b0
  sensor_component/external/sensor_component_description:
    type: git
    url: https://github.com/tier4/sensor_component_description.git
    version: a6a628186b124bc1af73c4f48a6dcd4b441391d3
  sensor_component/external/tamagawa_imu_driver:
    type: git
    url: https://github.com/tier4/tamagawa_imu_driver.git
    version: de4bf6be79aa2968cf2f62e0ebe1ff8a5797e6ad
  sensor_component/external/transport_drivers:
    type: git
    url: https://github.com/MapIV/transport_drivers.git
    version: d1e413482ca93c7618b005459a6a42e78e895442
  sensor_kit/external/awsim_sensor_kit_launch:
[... 74 lines cut ...]
[2:41](https://www.youtube.com/watch?v=QFD8jrXuS2Q&t=161s) End of test due to NPC traffic jam

### Test 2
- https://www.youtube.com/watch?v=NrlTcegIaCw
- Check the time tags in description for details

[1:38](https://www.youtube.com/watch?v=NrlTcegIaCw&t=98s) Test start
[2:04](https://www.youtube.com/watch?v=NrlTcegIaCw&t=124s) Random stop
[2:14](https://www.youtube.com/watch?v=NrlTcegIaCw&t=134s) Random stop
[2:29](https://www.youtube.com/watch?v=NrlTcegIaCw&t=149s) Attempts to change lane, stops unexpectedly
[2:56](https://www.youtube.com/watch?v=NrlTcegIaCw&t=176s) Fails to go back to the old lane
[3:07](https://www.youtube.com/watch?v=NrlTcegIaCw&t=187s) Stuck waiting in the middle of the road
[3:52](https://www.youtube.com/watch?v=NrlTcegIaCw&t=232s) Stuck in the middle of the road without anything blocking the path

### Test 3
- https://www.youtube.com/watch?v=N3LyPfmPuDI
- Check the time tags in description for details

[1:08](https://www.youtube.com/watch?v=N3LyPfmPuDI&t=68s) Test start
[1:19](https://www.youtube.com/watch?v=N3LyPfmPuDI&t=79s) Crashes to obviously incoming vehicle from right and stops after it happens
[1:24](https://www.youtube.com/watch?v=N3LyPfmPuDI&t=84s) Stops before vehicle in front and keeps blocking the road
[2:10](https://www.youtube.com/watch?v=N3LyPfmPuDI&t=130s) fails to turn the empty road curve
[2:20](https://www.youtube.com/watch?v=N3LyPfmPuDI&t=140s) fails to turn the empty road curve
[2:45](https://www.youtube.com/watch?v=N3LyPfmPuDI&t=165s) waits too long in walkway
[3:17](https://www.youtube.com/watch?v=N3LyPfmPuDI&t=197s) stuck in narrow node

WJaworskiRobotec, 2023-08-31:
@xmfcx 

I've made a lot of tests using AWSIM with 1 (top) and all 3 lidars on the Lexus Vehicle. I have no problem with performance,even on my laptop:
- CPU:  8 cores 11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz
- RAM: 64 GB
- GeForce RTX 3070 Mobile

Regarding strange behavior of Autoware observed in your comment, I have noticed that the output of segmentation differs betwen 1 and 3 lidars (check the screenshots below). It can be seen that with 1 lidar, there is no segmentation result on the ground, however with 3, there are segmented points (green rectangle), surprisingly created from points from the top lidar:

- 1 LIDAR:
[image]

- 3 LIDARs:
[image]

If you have any questions, I can provide AWSIM binaries, more details on test cases etc.

xmfcx, 2023-08-31:
@WJaworskiRobotec thanks for testing.

> I've made a lot of tests using AWSIM with 1 (top) and all 3 lidars on the Lexus Vehicle. I have no problem with performance,even on my laptop

Could you add [Graphy](https://assetstore.unity.com/packages/tools/gui/graphy-ultimate-fps-counter-stats-monitor-debugger-105778) to the AWSIM project (import it and move the prefab to the scene root) and also record the runs with OBS so we can see the performance too?
```bash
# obs installation
sudo add-apt-repository ppa:obsproject/obs-studio
sudo apt install obs-studio
```

I am not convinced about the performance, on my desktop 3070 AWSIM was running at ~10fps, evident from the videos. It might look like it runs fine but I don't think you had 60fps on your laptop.

About the ground removing issue, I think that's a perception ground segmentation problem from here on. It doesn't seem to be playing well with multiple lidars.

## Short summary of my new tests with multiple machines

Today I have made some more serious tests on 10Gbps connected double machine setup where Autoware running on 5900X and RTX3090 and AWSIM running on 5900X and RTX4080.

Some summary of my findings:

On awsim-stable branch, traffic light recognition works. I've made my new tests on this branch.

First I tried default single vlp16 lidar test.
It worked fine but perception was struggling to work fine with so little points obviously. (No ground segmentation problem.)

Then activated side lidars and rerun the tests. Performance became choppy, I think something is wrong the ROS2 bridge or the way the data is transmitted, maybe timestamped. Because the combined point cloud sometimes lacked some lidars flashing on and off, very annoying to witness. (Also the ground segmentation problem is present as expected.)

Then made the rest of my tests by turning off the side lidars and making the top lidar into a VLS128. It had 0.2 horizontal resolution (I think this is in degrees.) It didn't have the flashing lidars issue anymore but the performance dropped to 20-30 FPS.

Then I cranked down the resolution to 0.4 degrees and it was running at 40fps, I've found it to be ok. And made the rest of my tests there. With single lidar, ground segmentation worked fine. Also perception part had enough coverage to work with.

I will make a more comprehensive report and share my findings tomorrow.

WJaworskiRobotec, 2023-08-31:
@xmfcx 

Regarding the performance : in today's test it was not my main focus, I just wanted to investigate the problems with Autoware driving behavior.

Seems that we are aligned about the segmentation problem, so now we will focus on the performance evaluation. Unfortunately, I'm on a business trip next week so I cannot do it myself, but will assign someone from our team to continue this investigation.

msz-rai, 2023-09-04:
@xmfcx cc @WJaworskiRobotec 

Regarding point cloud flickering, we have found an issue in our LiDAR simulation plugin. The problem is that we synchronize a scene with our native library once per frame while LiDARs trigger in `FixedUpdate` of the simulation. `FixedUpdate` can be called multiple times per frame, if the frame rate is low. In this case, we are raytracing on the same scene changing only LiDARs position which may overlap with the body of the car.

We are working on a fix to this bug.

xmfcx, 2023-09-06:
I will share my new tests in here: https://github.com/orgs/autowarefoundation/discussions/3813

closed 2025-05-21 by Goldob