autowarefoundation/autoware-projects #70: Design of Autoware + AWSIM evaluation procedure

WJaworskiRobotec, 2023-09-26:


lchojnack, 2023-10-10:
I've prepared [a wiki page](https://github.com/autowarefoundation/autoware-projects/wiki/Autoware--AWSIM-verification) where I've started to describe how Autoware AWSIM verification can be done. 

For test execution, I think we can use [scenario_simulator_v2](https://github.com/tier4/scenario_simulator_v2). As I know, the support for executing scenarios using AWSIM is in progress and will be available in the near future. 

In order to think about and write down scenarios/test cases, I've prepared a [spreadsheet](https://docs.google.com/spreadsheets/d/1WqcU0fPepVea-U89tJd-lxoP8H_HLyHyvZ7JXfyqI_M/edit#gid=0). At the beginning we can try to reproduce the issues encountered in the previously performed tests ([in the description](https://github.com/autowarefoundation/autoware-projects/issues/70#issue-1913061680))

Please let me know your feedback

lchojnack, 2023-11-15:
I've started to prepare the AWSIM simulation which supports [scenario_simulator_v2](https://github.com/tier4/scenario_simulator_v2) scenarios.  
As a first step, I've used the ITRI scenario from [here](https://evaluation.tier4.jp/evaluation/reports/846bee80-98bf-5529-aae9-a30892e833f8/tests/691dc5af-c0d3-5df4-b9b9-ae0539e1f88f/211b91d5-c6f0-52a3-a650-0b96ab743403/893bd875-82bf-5339-9ad6-80373bfb8fc5?project_id=awf)

I've edited the scenario in the following way
```diff
@@ -31,6 +31,7 @@ OpenSCENARIO:
         Vehicle:
           name: ''
           vehicleCategory: car
+          model3d: "lexus_rx450h"
           BoundingBox:
             Center:
               x: 1.48
@@ -70,6 +71,7 @@ OpenSCENARIO:
         Vehicle:
           name: ''
           vehicleCategory: car
+          model3d: "taxi"
           BoundingBox:
             Center:
               x: 0
```
--- 
Then, I prepared the AWSIM with ITRI scene and ss2 support, and I ran the AWSIM with the scenario. The result is presented in the video below

https://github.com/autowarefoundation/autoware-projects/assets/8416175/9613682b-a646-475e-be14-938f05798dc6

---
At this moment, SS2 is able to spawn EGO and NPC and start the scenario execution. There is an issue that EGO is not able to detect the NPC and crash it. For that reason, the scenario fails. I'm investigating why the issue happens.

---
Tested with
| | |
| -- | --| 
|AWSIM       |       https://github.com/tier4/AWSIM/commit/df76d6f1d7cca89e834007bd5aa459ac0469cdf2 |
|Unitypackage|  `ITRI_Campus_r02_HDRP.unitypackage` |
|Autoware     |    https://github.com/autowarefoundation/autoware/tree/2023.10 |
|SS2                   | https://github.com/tier4/scenario_simulator_v2/commit/e187b8f7844c0c4464b3feff5680d66d8bbfde03|

lchojnack, 2023-12-19:
I've retested the scenario with an adjusted ITRI map to ll2. The scenario passes successfully.  In the next steps, I will test the prepared AWSIM binary with other scenarios.

https://github.com/autowarefoundation/autoware-projects/assets/8416175/be8ce7c7-5fe6-4f3b-8909-05249a2d3211

lchojnack, 2024-01-17:
I've prepared the AWSIM binary of ITRI scene which supports ss2 and modified openscenarios. There are available [here](https://drive.google.com/drive/folders/1m5HlympOei2N2mqMHnmoeb78iwv0mX6Z?usp=sharing)

lchojnack, 2024-02-14:
I've started to work on the testing scenarios. So far, I've prepared initial version of scenarios which verify the following planning features:
- [Lane Change](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-architecture/planning/#supported-features:~:text=Demonstration%20Video-,Lane%20Change,-Plan%20path%20for) 
- [Obstacle Avoidance](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-architecture/planning/#supported-features:~:text=map%20(driving%20lanelets)-,Obstacle%20Avoidance,-Plan%20path%20to)

So far, the scenarios are available on gdrive, but I think storing the scenarios in the GitHub repository could be a good idea.

### Lane Change
[scenario](https://drive.google.com/file/d/1KLYb6GPeHiioEYkHlSMxeHSZXUfuHmOa/view?usp=drive_link)

https://github.com/autowarefoundation/autoware-projects/assets/8416175/a59ac096-e4db-487c-8088-0f08496ab397

### Obstacle Avoidance
[scenario](https://drive.google.com/file/d/1bNbLbffzS6wpJxhzWPb05EDx3Uc9DPlp/view?usp=sharing) 

https://github.com/autowarefoundation/autoware-projects/assets/8416175/fcdd6381-4b7d-4d37-862f-85c97dfdbe68

lchojnack, 2024-03-20:
I've created one more scenario when a pedestrian walks on a crosswalk.

### PedestrianOnCrosswalk

[scenario](https://drive.google.com/file/d/1G4F-qpBzXNj4VzptoJamq0rN4xspU31Z/view?usp=drive_link)

https://github.com/autowarefoundation/autoware-projects/assets/8416175/b312bb27-eda9-45ea-aa31-72bfd0841e77

#### Conclusion

I've noticed that when the scenario is run with AWSIM, the EGO doesn't detect the NPC correctly and finally hits it.
When the scenario is run only with SS2, then the EGO stops before the pedestrian. I think the issue is related to Autoware perception pipeline.  

https://github.com/autowarefoundation/autoware-projects/assets/8416175/3b3d5412-82cc-47bf-ba4e-e10995f5a1ca

xmfcx, 2025-05-21:
This task is finished:
- https://tier4.github.io/AWSIM/GettingStarted/UsingOpenSCENARIO/

closed 2025-05-21 by Goldob